ethz-asl/rotors_simulator #737: Ubuntu 20.04 with ROS noetic catkin build issue

TianxiaoYee, 2024-10-21:
I am trying to install RotorS on my Ubuntu20.04+Ros noetic full desktop version. I have faced this issue for two weeks. It looks like it's about mavlink_status_t but I have absolutely no idea how to fix it.

Bug report (just a part):

"
In file included from /opt/ros/noetic/include/mavlink/v2.0/common/../message.hpp:33,
                 from /opt/ros/noetic/include/mavlink/v2.0/common/common.hpp:16,
                 from /home/ros20/Mavros_ws/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /home/ros20/Mavros_ws/src/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:23,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:1:
/opt/ros/noetic/include/mavlink/v2.0/common/../mavlink_helpers.h:224:31: note:   initializing argument 4 of ‘uint16_t mavlink::mavlink_finalize_message_buffer(mavlink::mavlink_message_t*, uint8_t, uint8_t, mavlink::mavlink_status_t*, uint8_t, uint8_t, uint8_t)’
  224 |             mavlink_status_t* status, uint8_t min_length, uint8_t length, uint8_t crc_extra)
      |             ~~~~~~~~~~~~~~~~~~^~~~~~
In file included from /opt/ros/noetic/include/mavlink/v2.0/common/common.h:2858,
                 from /opt/ros/noetic/include/mavlink/v2.0/common/mavlink.h:32,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:27,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:1:
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_debug.h: At global scope:
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_debug.h:188:97: error: ‘mavlink_status_t’ has not been declared
  188 | static inline uint16_t mavlink_msg_debug_encode_status(uint8_t system_id, uint8_t component_id, mavlink_status_* _status, mavlink_message_t* msg, const mavlink_debug_t* debug)
      |                                                                                                 ^~~~~~~~~~~~~~~
In file included from /opt/ros/noetic/include/mavlink/v2.0/common/common.h:2859,
                 from /opt/ros/noetic/include/mavlink/v2.0/common/mavlink.h:32,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:27,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:1:
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_setup_signing.h:95:103: error: ‘mavlink_status_t’ has not been declared
   95 | static inline uint16_t mavlink_msg_setup_signing_pack_status(uint8_t system_id, uint8_t component_id, mavlink_status_t *_status, mavlink_message_t* msg,
      |                                                                                                       ^~~~~~~~~~~~~~~~
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_setup_signing.h: In function ‘uint16_t mavlink_msg_setup_signing_pack_status(uint8_t, uint8_t, int*, mavlink_message_t*, uint8_t, uint8_t, const uint8_t*, uint64_t)’:
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_setup_signing.h:116:74: error: cannot convert ‘int*’ to ‘mavlink::mavlink_status_t*’ {aka ‘mavlink::__mavlink_status*’}
  116 |     return mavlink_finalize_message_buffer(msg, system_id, component_id, _status, MAVLINK_MSG_ID_SETUP_SIGNING_MIN_LEN, MAVLINK_MSG_ID_SETUP_SIGNING_LEN, MAVLINK_MSG_ID_SETUP_SIGNING_CRC);
      |                                                                          ^~~~~~~
      |                                                                          |
      |                                                                          int*
In file included from /opt/ros/noetic/include/mavlink/v2.0/common/../message.hpp:33,
                 from /opt/ros/noetic/include/mavlink/v2.0/common/common.hpp:16,
                 from /home/ros20/Mavros_ws/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /home/ros20/Mavros_ws/src/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:23,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:1:
/opt/ros/noetic/include/mavlink/v2.0/common/../mavlink_helpers.h:224:31: note:   initializing argument 4 of ‘uint16_t mavlink::mavlink_finalize_message_buffer(mavlink::mavlink_message_t*, uint8_t, uint8_t, mavlink::mavlink_status_t*, uint8_t, uint8_t, uint8_t)’
  224 |             mavlink_status_t* status, uint8_t min_length, uint8_t length, uint8_t crc_extra)
      |             ~~~~~~~~~~~~~~~~~~^~~~~~
In file included from /opt/ros/noetic/include/mavlink/v2.0/common/common.h:2859,
                 from /opt/ros/noetic/include/mavlink/v2.0/common/mavlink.h:32,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:27,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:1:
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_setup_signing.h: At global scope:
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_setup_signing.h:194:105: error: ‘mavlink_status_t’ has not been declared
  194 | static inline uint16_t mavlink_msg_setup_signing_encode_status(uint8_t system_id, uint8_t component_id, mavlink_status_t* _status, mavlink_message_t* msg, const mavlink_setup_signing_t* setup_signing)
      |                                                                                                         ^~~~~~~~~~~~~~~~
In file included from /opt/ros/noetic/include/mavlink/v2.0/common/common.h:2860,
                 from /opt/ros/noetic/include/mavlink/v2.0/common/mavlink.h:32,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:27,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/src/hil_sensor_level_interface.cpp:1:
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_button_change.h:90:103: error: ‘mavlink_status_t’ has not been declared
   90 | static inline uint16_t mavlink_msg_button_change_pack_status(uint8_t system_id, uint8_t component_id, mavlink_status_t *_status, mavlink_message_t* msg,
      |                                                                                                       ^~~~~~~~~~~~~~~~
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_button_change.h: In function ‘uint16_t mavlink_msg_button_change_pack_status(uint8_t, uint8_t, int*, mavlink_message_t*, uint32_t, uint32_t, uint8_t)’:
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_button_change.h:111:74: error: cannot convert ‘int*’ to ‘mavlink::mavlink_status_t*’ {aka ‘mavlink::__mavlink_status*’}
  111 |     return mavlink_finalize_message_buffer(msg, system_id, component_id, _status, MAVLINK_MSG_ID_BUTTON_CHANGE_MIN_LEN, MAVLINK_MSG_ID_BUTTON_CHANGE_LEN, MAVLINK_MSG_ID_BUTTON_CHANGE_CRC);
      |                                                                          ^~~~~~~
      |                                                                          |
      |                                                                          int*
In file included from /opt/ros/noetic/include/mavlink/v2.0/common/../message.hpp:33,
                 from /opt/ros/noetic/include/mavlink/v2.0/common/common.hpp:16,
                 from /home/ros20/Mavros_ws/src/mavros/libmavconn/include/mavconn/mavlink_dialect.h:26,
                 from /home/ros20/Mavros_ws/src/mavros/mavros_msgs/include/mavros_msgs/mavlink_convert.h:18,
                 from /home/ros20/RotorS/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h:23,
[... 385 lines cut ...]
      |                                                                                                             ^~~~~~~~~~~~~~~~
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_video_stream_status.h: In function ‘uint16_t mavlink_msg_video_stream_status_pack_status(uint8_t, uint8_t, int*, mavlink_message_t*, uint8_t, uint16_t, float, uint16_t, uint16_t, uint32_t, uint [... cut ...]
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_video_stream_status.h: At global scope:
/opt/ros/noetic/include/mavlink/v2.0/common/./mavlink_msg_video_stream_status.h:260:111: error: ‘mavlink_status_t’ has not been declared
  260 | static inline uint16_t mavlink_msg_video_stream_status_encode_status(uint8_t system_id, uint8_t component_id, mavlink_status_t* _status, mavlink_message_t* msg, const mavlink_video_stream_status_t* video_stream_status)
"
Any suggestions would be highly appreciated.

hasankiyani007, 2024-10-22:
Same exact issue. Recently had to clean my catkin workspace, and while rebuilding, im getting this exact same issue. Same environment as well. Any suggestions?

MouDARLEK, 2024-10-25:
Been here, add this may help.
[image]

MouDARLEK, 2024-10-25:
[image]
add this may help

hasankiyani007, 2024-10-25:
That worked! Just for info, how exactly did you come across that specific solution? I know that the error says mavlink status hasn't been declared, but how did you know to specifically add this into that specific file and not somewhere else?

HuaYuXiao, 2024-12-08:
> <img alt="Snipaste_2024-10-25_16-24-38" width="561" src="https://private-user-images.githubusercontent.com/104193537/380075803-bb2003c8-3809-4db0-ad21-8ebc9558314a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM2Mzk1NzcsIm5iZiI6MTczMzYzOTI3NywicGF0aCI6Ii8xMDQxOTM1MzcvMzgwMDc1ODAzLWJiMjAwM2M4LTM4MDktNGRiMC1hZDIxLThlYmM5NTU4MzE0YS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOFQwNjI3NTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03NGEwNmUyZjcwY2UwMzI5ODk4MGI1NDg1MDc0MWQ0YmZlYmQyMTQ3OGIxYzU3NGVmZTU0ZGEzODQ2MTM4ZTNhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.LXEhFkix7ET9Ekr-Opt_JrjL5rZ2Z5Q2zAK6f8bLybs"> add this may help

Thank U! This worked for me!

LoudLove, 2025-03-31:
> <img alt="Snipaste_2024-10-25_16-24-38" width="561" [...]

Your assistance would be immensely appreciated！

Wingflysky, 2025-07-24:
Thank you for your solution - it worked perfectly!
Compilation Error
error: 'mavlink_status_t' has not been declared
Solution
Add the missing type definition in the correct location:

​​File​​: ws/src/rotors_simulator/rotors_hil_interface/include/rotors_hil_interface/hil_interface.h

​​Modification​​:
#ifndef MAVLINK_H
  typedef mavlink::mavlink_message_t mavlink_message_t;
  // ▼▼▼ Add this critical line ▼▼▼
  typedef mavlink::mavlink_status_t mavlink_status_t;
  // ▲▲▲ Solves 'has not been declared' error ▲▲▲
  #include <mavlink/v2.0/common/mavlink.h>
#endif

RanFR, 2025-07-29:
Thank you so much!!! It works well!!!

> <img alt="Snipaste_2024-10-25_16-24-38" width="561" [...]